petzval/btferret #2: Huge delay for several continuous packets

Akshatasutar, 2022-03-07:
Hi petzval, 
I have been using this library for the past week and it is great! The documentation is so clear. But I am running into a problem. I'm communicating between two Raspberry pis, one acting as a LE node server, that listens and receives packets, and the other as a node client, that sends packets to the node server continuously, every 500ms, until the loop is exited through a keyboard press. I'm using the function _write_node()_ at the client to send the data. This data is just a 3 byte array of chars. But after a while, after the first 300 messages, the transfer lags a lot. By a few seconds. This is a huge delay, considering I am sending values to control the speed of a motor. How to I reduce this huge delay?

petzval, 2022-03-07:
I've set up the same process as follows, but it seems OK. Is this 
essentially your code?

CLIENT CODE, node 1
connect_node(2,CHANNEL_NODE,0);
for(n = 0 ; n < 1000 ; ++n)
    {
   dat[0] = 12;  // must not be 10
   dat[1] = 34;  // must not be 10
   dat[2] = 10;    // always = end packet char
   write_node(2,dat,3);
   // 500ms delay
   }
disconnect_node(2);

SERVER CODE, node 2
node_server(1,callback,10);    // end packet char=10

int callback(int clientnode,unsigned char *dat,int count)
   {
   // count should be 3
   // code to use information in dat[0]/dat[1]
   return(SERVER_CONTINUE);
   }

Akshatasutar, 2022-03-07:
Yes. That is essentially it. I print out the value sent on the sending pi, and the value received on the receiving pi, as well as the packet number. There seems to be a delay between sending and receiving.

Akshatasutar, 2022-03-08:
Hi. Just wanted to let you know it was a mistake from my side regarding the delay functions. It was basically producing no delay at all. It is fixed now.

Akshatasutar, 2022-03-09:
Hi again, 
I don't know if I should create a seperate issue for this. I have the same setup as before - two LE pis acting as node server and client. Now, I want to get a sort of notification on the client side when the server sends an acknowledgement. I mean I want to track when replies with opcode 13 are made. From what I understood, the read functions only read when the server writes to the client. Am I missing something? Is there a function to, say, return 1 when an ack has been received? I went through the hciread() function, but it was difficult for me to use that in my main function.

petzval, 2022-03-10:
By opcode 13, do you mean the HCI_Number_Of_Completed_Packets 0x13 Event 
sent by the Bluetooth controller, or a data packet programmed in C to be 
sent by the node server to the client containing that custom opcode?

Akshatasutar, 2022-03-10:
I mean the HCI packet, 0x13 Event, yes. 
I have figured out a way now. I take the verbose print output file, that displays all the HCI and L2CAP packets, and count the lines using a python script starting with "> Event 13", to get the number of ACKS received. Thanks a ton for the verbose print mode, and a detailed description of the packets. 

But if you have a better way, let me know. Otherwise, you can ignore this and carry on with your day!

petzval, 2022-03-10:
Here's another way, but it means changing btlib.c and will have to be 
repeated with future versions.

Define a global variable (say, globalflag) that is visible to the client 
code and btlib.c.
Make readhci() visible to the client code. Modify readhci() as follows:

EXISTING code in readhci()
....
     else if( (sflag & IN_CNAME) != 0 && buf[1] == 0x07 && buf[3] == 0 
)  // status 0
       {
       gotflag = IN_CNAME;
       n0 = 4;
       }
....
ADD this after above existing code
     else if(buf[1] == 0x13)
       globalflag = 1;

CLIENT CODE to wait for opcode 13

globalflag = 0;

   // do operation that should trigger opcode 13

   // infinite wait for opcode 13
   // globalflag may already be 1

while(globalflag == 0)
   readhci(0,0,0,100,0);    // runs for 100ms

Akshatasutar, 2022-03-10:
Thanks for this!