petzval/btferret #8: Bug by read_ctic

Ectalite, 2022-10-11:
I found a bug by calling the read_ctic function.
After trying out multiple times, I found out that reading 1, 2 and 4 bytes does work but not 3.
(For my particular case I am sending a 3 byte long information for rgb colors).

He is the code I'm using:
```
typedef struct sColor
{
  uint8_t r;
  uint8_t g;
  uint8_t b;
} sColor_t;

//Some more code...

char cBuffer[5];
iError = read_ctic(localnode(),3,cBuffer,3);
if(ERROR_FATAL != iError)
{
   sChoosedColor.r = (uint8_t)cBuffer[0];
   sChoosedColor.g = (uint8_t)cBuffer[1];
   sChoosedColor.b = (uint8_t)cBuffer[2];
   printf("sChoosedColor = r:%d g:%d b:%d\n",sChoosedColor.r,sChoosedColor.g,sChoosedColor.b);
}
else
{
   printf("Fatal error by Color reading\n");
}
```

petzval, 2022-10-11:
read_ctic returns the number of bytes read, not the error code. The value of ERROR_FATAL happens to be 3. If there is an error in read_ctic, it will print a message describing the problem.

```
if(read_ctic(...) == 0)
  {
  error = read_error();
  if(error == ERROR_FATAL)
    ....
```

Ectalite, 2022-10-11:
Oh no, I'm sorry, I missread the README, because at https://github.com/petzval/btferret#4-2-28-read_ctic it is written that read_error() gives out ERROR_FATAL. 
But yeah, read_ctic is not read_error()
So it's not a bug!

closed 2022-10-11 by Ectalite