petzval/btferret #11: LE and Classic server

LasseTS, 2023-01-04:
Hi,
Thank you for sharing this extremely useful tool!
Is there a way to run both a LE and Classic server at the same time on the same RPI device? 
Will I have to start to independently threaded instances of LE_Server and Node_Server?
Many Thanks!

petzval, 2023-01-05:
As things stand, it is not possible to run two servers simultaneously. I've had one report of multiple threads working, but I have not tried it myself, so I cannot say if it will work for two servers. I think you will need two separate programs that open separate HCI sockets. (Note - a Classic server and a Node server are different things. A Node server is a raw LE connection).

gregoiregentil, 2023-03-08:
+1. I'm really interested by this case. I have read the code and there are a lot of static variables. I'm wondering whether there could create some problems in case of two threads in the same executable, one for classic, one for ble.

Is there any recommendation here? Having external processes would make it very complicated in my case.

closed 2024-03-18 by petzval