petzval/btferret #12: notification callbacks get only 20 bytes

wolfk-wk, 2023-01-10:
I want to use your nice C package to connect from my PC to a BLE chessboard (Chessnut Air). All communications to the board  work well, except the notifiaction callbacks always receive only 20 bytes from the actually sent 38 bytes by the board.
I am sure that the board is OK, since it works well with an USB cable & Android apps.

Some lines of my code:
  cticn = 5;  
  int flag = 1; // inputint("0=Disable 1=Enable");
  char *s = "enabled";
  if (ctic_ok(node,cticn) == 0)    {   printf("Invalid index\n");   return 1;  }
  if (notify_ctic(node,cticn,flag, conf_callback) == 0) s = "failed";
  printf("notifyle %s %s %s cticn=%d\n",device_name(node),ctic_name(node,cticn),s,cticn);

  cticn = 3;
  if (notify_ctic(node,cticn,flag,board_notify_callback) == 0) s = "failed";
  printf("notifyle %s %s %s cticn=%d\n",device_name(node),ctic_name(node,cticn),s,cticn);

petzval, 2023-01-10:
Yes, I've been updating the code to increase the maximum characteristic size from the default 20. Today I uploaded a new version 8 of btlib.c that allows a maximum size of 244. Try the new version, and let me know if it doesn't work because this is an issue that it is intended to fix.

wolfk-wk, 2023-01-10:
Hello Petzval,
I have downloaded your new version 8.
It works now perfectly, the callback gets all the expected 38 bytes ;-)
Thanks for your work!

petzval, 2023-01-11:
Thanks for letting me know. I have just noticed a node server bug in version 8, so I have uploaded a new btlib.c to fix that.

closed 2024-03-18 by petzval