petzval/btferret #13: read_notify consumes 100% CPU

wolfk-wk, 2023-01-13:
When I call 'read_notify' in my C-program, after all initialisations to the device have been properly done, it consumes 100% of CPU time of my Ubuntu-Notebook (1 thread is completely busy). Easy to see with 'top'.
The same effect occurs with the test program btferret.

Is there any way to prevent this behaviour?

petzval, 2023-01-14:
The code is not intended for a multitasking system, it is single-thread synchronous, so nothing runs in the background. This makes the source code fairly straightforward and easy for users to modify. The read_notify and server codes are just running a continuous loop waiting for input from Bluetooth, and so tie up the system. However, the read_notify function does not have to be running when a notification is sent. The Bluetooth system stores notifications in its buffer until read_notify reads them. So it is possible to poll for notifications without missing one, and maybe there is a delay/sleep method that frees the CPU when running.

```
  do
    {
    // delay method that does not consume CPU
    read_notify(0);     // 0 ms wait = peek for any stored notifications
    }
  while......
```

Or in a multitasking environment like Windows, start a timer to poll for notifications.

```
  // WM_TIMER message from OS

  read_notify(0);

  // return to OS
```

wolfk-wk, 2023-01-16:
My solution is similar:
   for (int nloop;; nloop++) { //
       readhci(0,0,0,0,0); usleep(100*1000); //  = 100 msec
       .... //other code
    }
It works for me, since the average send rate from the device is approx 10/sec.
The CPU usage drops signifcantly.

petzval, 2023-01-17:
Thanks for letting me know, that's useful information.

closed 2024-03-18 by petzval