petzval/btferret #14: Save addresses in device.txt

wolfk-wk, 2023-01-17:
I have problems with saving found LE-devices in the file 'device.txt'. First I do a scan of LE-devices with btferret. First I do a LE-scan with btferret:
'> b'
```
.... This is my device:

3 FOUND 00:1B:10:4F:FFFFF - Fixed
    Flags = 06
    Manuf specific = 50 44 43 53 CB 54 77 64 00 00 B8 AD 4F 10 1B 00
    Tx Power Level = 0A
    No name so set = LE node 1003
    New device LE node 1003
....
```
Then connect to node 1003 and all the rest works correctly.
But if I try to save the device in 'devices.txt' as
```
DEVICE = Chessnut  TYPE=LE node=1 ADDRESS =  00:1B:10:4F:FFFFF
```
I get an format error:
```
Device data from devices.txt file
DEVICE = Chessnut  TYPE=LE node=1 ADDRESS =  00:1B:10:4F:FFFFF
Hex format error - odd number of digits
   ERROR **** Device address must be 6 bytes
************ initblue() FAILED ************
```
So I have to scan for LE-devices each time at program start, which is a little bit annoying🙁.
Can you tell me what would be the correct entry for this?

petzval, 2023-01-18:
This looks like a difference between the Raspberry Pi C compiler/library and the one you are using because I cannot reproduce it on a Pi with gcc. The address should be 6 bytes separated by 5 colons - 11:22:33:44:55:66. The FFFF value appears because the compiler is treating that number as a signed byte rather than unsigned. Try this fix. If it works, let me know and I'll update the code to account for other compilers.

```
In btlib.c - change char to unsigned char as follows:

Line 363
char *baddstr(unsigned char *badd,int dirn);

Line 2003
char *baddstr(unsigned char *badd,int dirn)

And if that does not work:

Line 2014
sprintf(s+3*n,"%02X:",(unsigned char)badd[k]);

```

wolfk-wk, 2023-01-18:
I modified only line 2014 in btlib.c.  Now it seems to work with my gcc : gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1) 
 
This is the new output of LE-scan:
```
> b
Scanning for LE devices - 10 seconds..
...
2 FOUND 00:1B:10:4F:AD:B8 - Fixed
    Flags = 06
    Manuf specific = 50 44 43 53 CB 54 77 64 00 00 B8 AD 4F 10 1B 00
    Tx Power Level = 0A
    No name so set = LE node 1002
    New device LE node 1002
....
```
After adding the address  00:1B:10:4F:AD:B8 to devices.txt all works perfectly. Thanks again!

petzval, 2023-01-18:
OK, thanks. Seems like inconsistent sprintf behaviour.

closed 2024-03-18 by petzval