petzval/btferret #15: Handling of server disconnections

wolfk-wk, 2023-01-22:
Sometimes the LE-server of my chessnut device disconnects (in case of weak connections). 
Then I want to start a new connection, but my client thread gets stuck in function 'readhci()' and so no further actions are  possible.
(Anyway, every 10 seconds I see a message '... not  connected' from btlib.c).
This is the output of btlib.c:
```
...
 4999  6234 board_cb received 38 byte
chessnut has disconnected
chessnut not connected as LE server
chessnut not connected as LE server
...
```
How is it possible to react to this disconnections?

petzval, 2023-01-22:
The readhci function should never get stuck, it will always return after the specified time out. I suspect the problem is with the handling of the two long long parameters. Use readhci directly in your code as follows, and use device_connected to check for disconnections:

```
extern int readhci(int,long long int,long long int,int,int);

do
  {
  readhci(0,(long long int)0,(long long int)0,0,0);
  if(device_connected(node) == 0)
    // has disconnected
  }

```

wolfk-wk, 2023-01-22:
Thanks. The two 'long long int' were already in the declaration, and the explicit cast in the function call  is not necessary. In my code I testet the return value  of readhci == 0, but this did not work. The next call to readhci obviously blocks.
However, your proposal 
```
 if (device_connected(node) == 0)
```
works perfectly.