petzval/btferret #16: Multiple BLE Services and Change Current One

crsawyer45, 2023-02-15:
Hey, BLE noobie here...I don't see a way to add more than one BLE service using either the documented methods or the Devices.txt file.  I would also like to change the one that is used by default to something else on the fly. Is this a possibility and I am just missing something?
I have the multiple characteristics within the default service seemingly working great. Thanks

petzval, 2023-02-16:
The BLE services are defined in the devices.txt file, so you set them by writing your own devices file using the instructions in documentation sections 3.3 (Devices file) and 4.2.17 (LE server). Delete all the default service LECHAR entries and add any number of your own. Once the services have been set up (when init_blue reads the devices file) they cannot be added to, deleted or changed via function calls. This is because many LE clients cannot cope with changing services. They read the services when they first connect, and then assume that they remain unchanged. Some clients even store the information, and after disconnection and re-connection they assume that the services are still the same, and do not read them a second time. (See SERVICES CHANGED in section 3.7 LE server).

crsawyer45, 2023-02-17:
Okay, i have read the documentation and looked at the examples...A value of 112233445566778899AABBCCDDEEFFFF is set as teh service UUID. I am able to change the UUID's of the characteristics within the service, but I would also like to change the UUID of the service from the above to something else and also potentially add more than just the one service.  Is this possible and configurable from within the devices.txt file or is there some other way I may easily do this?

petzval, 2023-02-18:
I now see that it is a primary service question. You can change the primary service UUID, but only by editing the btlib.c file. The UUID is hard coded as baseuuid[] and is at around line 669. It is not possible to add more primary services.

closed 2024-03-18 by petzval