petzval/btferret #20: Using the library as "daemon" without stdin

gregoiregentil, 2023-03-08:
I'm trying to integrate this library in an application that already uses stdin. It would be great to add a daemon option to the library so that there is no conflict about stdin.

Accessory, I would be interested to know how I should patch btlib.c to prevent stdin usage. Should I return 0 in setkeymode and readkey?

petzval, 2023-03-08:
STDIN is only used for three things: 
1. Input PIN code. This can be avoided by putting the PIN code in devices.txt, and the code around line 6176 that prompts for the code can be deleted.
2. Input passkey. This can be avoided by using security that does not require passkey input from the btferret keyboard, and the code around line 6207 that prompts for input can be deleted.
3. Exit from servers by pressing x key. As you say, this can be disabled by returning 0 from setkeymode and readkey. But you will have to rely on SERVER_EXIT returns to stop a server, or find some other way of stopping a server that is stuck waiting for input.

gregoiregentil, 2023-03-08:
This is very useful. Thank you. A second argument to init_blue_ex noinput to init would be the best.

init_blue_ex(char *filename,int hcin, int noinput);

closed 2024-03-18 by petzval