petzval/btferret #21: How to modify BLE mininterval,maxinterval and supervision?

gregoiregentil, 2023-03-08:
During advertisement of a BLE server, the peripheral can suggest some values for min-interval, max-interval and supervision to the central. The central who usually plays the role of client decides in the end. It would be great to be able to modify those values especially the recommending ones on server side?

gregoiregentil, 2023-03-08:
Perhaps, if it's too complicated to expose in devices.txt, you could point out where it is in the code. Note that those values are passed to the Linux kernel which can have it's default supervision value (/sys/kernel/debug/bluetooth/hci0/supervision_timeout).

petzval, 2023-03-08:
The connection and supervision values are not in the advertising parameters command (OGF 08 OCF 06) and as far as I can see there is no advertising data code for them (OGF 08 OCF 08). They are in the client's connection command (OGF 08 OCF 0D) that is implemented via leopen[] at line 497.  There is a connection update command (OGF 08 OCF 13) that the client or peripheral can issue to change the parameters after connection as follows:

```
unsigned char conupdate[32] = { 18,0,S2_HAND,0,0x01,0x13,0x20,0x0E,0,0,0,0,0,0,0,0,0,0,0,0,0,0};

// 14 parameters defined in Vol 4 PtE 7.8.18

sendhci(conupdate,ndevice);
```

gregoiregentil, 2023-03-10:
I have read volume 4. That is correct. I will potentially play with connupdate. Thank you for the clarification.

closed 2023-03-10 by gregoiregentil