petzval/btferret #22: Can't make work notification

gregoiregentil, 2023-03-09:
I have a hard-time to make work notification. Here is a simple case that doesn't work: https://gentil.com/tmp/test.zip

petzval, 2023-03-09:
The notification process normally works as follows:
1, The btferret server specifies a characteristic with notify permission in devices.txt.
2. The btferret test.c code starts a server via le_server
3. Android connects to the server.
4. Android client enables the notification - not test.c
5. The btferret test.c code writes a value to the characteristic - Android must be connected
6. When a characteristic is written, btferret sends a notification to Android - not to read notify in test.c

gregoiregentil, 2023-03-10:
I was confused by read_notify which is for the client. I can make work notification though performance is not good (opening a different bug for that).

closed 2023-03-10 by gregoiregentil