petzval/btferret #24: Version 10 breaks best connection interval

gregoiregentil, 2023-04-14:
Those are Android logs. Note that my Android app requests the high connection priority.

Version 9 (with //paramreply[PAKHEADSIZE+j+6] = insdat[n+j+3];) ends in 7.5ms:
```
2023-04-13 20:42:37.331 32740-405   bt_stack                pid-32740                            I  [INFO:le_scanning_manager.cc(318)] SetScanParameters: scan_type=1, scan_interval=6553, scan_window=6553
2023-04-13 20:42:37.495 32740-405   bt_stack                pid-32740                            I  [INFO:le_scanning_manager.cc(318)] SetScanParameters: scan_type=1, scan_interval=2977, scan_window=324
2023-04-13 20:42:39.444  9443-9576  BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=6 latency=0 timeout=500 status=0
2023-04-13 20:42:39.599  9443-9576  BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=6 latency=0 timeout=500 status=59
2023-04-13 20:42:39.722  9443-9443  Capacitor/Console       sport.inout.inout                    I  File: http://localhost/main.js - Line 377 - Msg: Request HIGH ConnectionPriority for interval
2023-04-13 20:42:39.749  9443-9458  BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=6 latency=0 timeout=500 status=30
```

Version 10 ends in 45ms:
```
2023-04-13 20:38:08.310 32740-405   bt_stack                pid-32740                            I  [INFO:le_scanning_manager.cc(318)] SetScanParameters: scan_type=1, scan_interval=6553, scan_window=6553
2023-04-13 20:38:08.449 32740-405   bt_stack                pid-32740                            I  [INFO:le_scanning_manager.cc(318)] SetScanParameters: scan_type=1, scan_interval=2977, scan_window=324
2023-04-13 20:38:09.445  9443-9559  BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=6 latency=0 timeout=500 status=0
2023-04-13 20:38:09.667  9443-9559  BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=36 latency=0 timeout=500 status=0
2023-04-13 20:38:09.773  9443-9443  Capacitor/Console       sport.inout.inout                    I  File: http://localhost/main.js - Line 377 - Msg: Request HIGH ConnectionPriority for interval
2023-04-13 20:38:09.850  9443-9576  BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=36 latency=0 timeout=500 status=30
```

Version 10 + "if(op == LE_CONNECT) set_le_interval_server(clientnode,6,6);" ends in 11.25ms:
```
2023-04-13 20:50:55.174 32740-405   bt_stack                pid-32740                            I  [INFO:le_scanning_manager.cc(318)] SetScanParameters: scan_type=1, scan_interval=6553, scan_window=6553
2023-04-13 20:50:55.414 32740-405   bt_stack                pid-32740                            I  [INFO:le_scanning_manager.cc(318)] SetScanParameters: scan_type=1, scan_interval=2977, scan_window=324
2023-04-13 20:50:56.419  9443-9459  BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=9 latency=0 timeout=500 status=0
2023-04-13 20:50:56.598  9443-9459  BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=6 latency=0 timeout=500 status=0
2023-04-13 20:50:56.721  9443-10133 BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=9 latency=0 timeout=500 status=0
2023-04-13 20:50:56.776  9443-9443  Capacitor/Console       sport.inout.inout                    I  File: http://localhost/main.js - Line 377 - Msg: Request HIGH ConnectionPriority for interval
2023-04-13 20:50:56.801  9443-10133 BluetoothGatt           sport.inout.inout                    D  onConnectionUpdated() - Device=702C1F_A interval=9 latency=0 timeout=500 status=30
```

How can I get back my 7.5ms?

gregoiregentil, 2023-04-14:
//paramreply[PAKHEADSIZE+j+6] = insdat[n+j+3]; "restores" 7.5ms but I'm confused about the situation.

petzval, 2023-04-14:
First - there was a bug in the LE interval code that may or may not be relevant. A fixed version 10 has just been uploaded.
1. On my devices, Android high priority interval is 11.25 (interval 9)  to 15ms, so calling this will set to 9.
2. Calling set_le_interval_server(node,6,6) sets the interval to 9 - Android decides this and does not allow 6.
3. To my surprise, on my Android tablet running nRF connect, I can call set_le_interval_update(node,6,6) from the server and it works - Android allows 6.
4. Android sets the interval to 6 to read the services, but then restores it to a slower value. When it changes the interval it asks for a paramreply and the paramreply[PAKHEADSIZE+j+6] = insdat[n+j+3] code replies that the server agrees to the change by just copying the requested values to the reply. Removing this copy code returns the default reply which is coded as interval = 6. But if the Android has requested a slower interval, this is an illegal reply.  The server can only agree to values within the requested min-max range. I guess an illegal reply is ignored or results in no interval change.

Conclusion. Do not set the Android high priority, it will be 9. There are two possible solutions to get interval 6.

1. Call set_le_interval_update if it works, but do not call from LE_CONNECT because the Android may still be reading the services and will reset to a slower value when it is finished. Use an LE_READ or LE_WRITE to trigger the call when you are sure Android has finished the connection procedure.

```
if(op == LE_READ)
  set_le_interval_update(clientnode,6,6);
```
2. A messy solution. Let Android set the interval to 6 when it reads the services, but then reject any further changes by sending a negative reply if the requested interval is not 6. So at line 6761

```
else if(gotflag == IN_PARAMREQ)
  {
  if(insdat[n+3] == 6 && insdat[n+4] == 0)
    sendhci(paramreply,devicen);    // default reply min=6 max=6 
  else
    sendhci(paramnegreply,devicen);   // reject
  }
```

gregoiregentil, 2023-04-14:
OK. Thank you.

closed 2023-04-14 by gregoiregentil