petzval/btferret #25: Optimization to communicate with an iOS client

gregoiregentil, 2023-04-14:
I'm interested to improve the communication between btferret peripheral GATT server and iOS. You might want to read this page https://punchthrough.com/maximizing-ble-throughput-on-ios-and-android/ for interesting information.

Right now I see: https://pastebin.com/e0T1s95H. I think that 30ms makes sense considering the link above. How can I put btferret in "HID device" GATT server mode?

gregoiregentil, 2023-04-14:
There is more information about HID BLE here: https://novelbits.io/bluetooth-hid-devices-an-intro/

That doesn't seem "obvious". I don't think that btferret supports it for the moment.

For my project, I'm not interested to get a full BLE HID-compliant peripheral. I'm "only" interested to include a "fake" HID GATT service so that iOS accepts 11.25ms interval.

petzval, 2023-04-18:
It is not simple to make btferret look like an HID device, so there is no quick solution there.

closed 2024-03-18 by petzval

nmattia, 2025-10-06:
> HID had not been implemented at the time, but now it is.

Thanks for the context! I've created a new ticket: https://github.com/petzval/btferret/issues/68