petzval/btferret #28: Using mutliple adapters

r-bt, 2023-07-18:
I'm trying to connect to 15 BLE devices by splitting up the connections between three adapters. I can see that I should use `init_blue_ex` to specify the `hci` number of the adapter but this seems to set some global state. Is there a way that I can initialize three separate adapters such that I can work with all of them?

r-bt, 2023-07-27:
Looking through the code I can't see a way of doing this right now. This is the only interface I've found that is giving me reliable results so I'm going to try wrap some of the code in a c++ class to allow multiple adapters to be used at once here: https://github.com/r-bt/btbeaver. 

I was wondering if you have any notes on the code in `btlib` specifically which might help while I do this?

petzval, 2023-07-28:
The obvious approach is to turn btlib.c into an adapter class. So all the global variables (except the text strings and hci packets) will be in the class and each adapter has its own set. The operation of the static variable initflag in init_blue_ex will have to be taken into account - it is there to allow multiple calls to init_blue_ex for a single adapter if it is disabled and then re-enabled. I'd be interested to know how many connections btlib can handle through one adapter.

closed 2024-03-18 by petzval