petzval/btferret #31: Second run issue

bvgvova, 2023-11-20:
Hi!
Thank you for sharing this very useful library & brilliant explanation (readme).
Nice low-level approach.
But I encountered some problems, main is that after first run (>b ; >g) , PC (Ubuntu16) where program is running lost  BT-mouse.
​​​​​​​Attempting to run a btferrent once more leads to : 
    Unknown device 52:24:94:23:C7:A5 connected - rejecting.. // that is my mouse
    Unable to read local board address
    Device data from devices.txt file

I suppose that is because Bluez — «off».
ioctl with HCIDEVUP before exit has no effect.
I run vers.12. Any ideas?

petzval, 2023-11-21:
Thanks for your comments. You are right about bluez. It must be shut down or btferret will not work - they will compete with each other for access to the Bluetooth adapter. When btferret exits, the operating system should re-start bluez automatically. I guess that bluez cannot re-make the connection to the mouse when it re-starts. The "Unknown device.." message is caused because the mouse is trying to make a Classic connection. That seems to cause a problem because you should not get the "Unable to read local.." message. I'll have a look at it. I can only suggest disconnecting the mouse, run btferret, and after exit re-connect the mouse.

bvgvova, 2023-11-21:
Thank you for reply.
Sorry, I was wrong saying that ioctl have no effect. ( I simply forgot to firstly close_all() ) . So ioctl + close_all() solve a problem.
I am still digging in your code, so if you do not object, I'll ask you some more.
1. what means : if(strcmp(filename,"__BTLE__") == 0) [btlib .c 1259] .....( filename = "devices.txt" -- you bypassed init_btle()...)
2. why do you said that we shouldn't wait for a high speed from btferret ?  Is yours low-level approach is worse than  with Dbus? (I tried an example with gattlib - it is too slow in scanning. ) 
What I am attempting to do is BLE gateway for beackons, so scanning & reading adv. payload speed should be high. What do you think about using your lib as a base for it ? 
(sometimes I get segfaults, but that needs more tests & I'll do them)
BR

petzval, 2023-11-22:
The BTLE functions are for BeetleIN and are not used here. The speed comments apply to the serial data rate, not the speed of Bluetooth procedures. The speed of scanning will depend on how often the beacons transmit - not on the reading speed. You will have to modify lescanx() in btlib.c to extract beacon data.

closed 2024-03-18 by petzval