petzval/btferret #33: Compile for Windows and macOS?

ronaaron, 2023-12-10:
Hello -- 
I just stumbled across your excellent library, and I'm curious to try to compile as well for macOS and Windows (using mingw).

I think it should be possible, with some tweaks. Has anyone tried either of these?

petzval, 2023-12-10:
I have not heard of any attempts to port to mac or Windows. There is one thing to check before attempting it. Linux allows direct access to the Bluetooth adapter via HCI_CHANNEL_USER. See the bind operation at around line 8190 in hcisock() function in btlib.c. If you cannot open a socket and bind to HCI_CHANNEL_USER then btferret cannot work. I think I have tried this on Windows 10 and it failed. If it does work, let me know and I'll think about the linux-specific functions that may cause problems.

ronaaron, 2023-12-10:
Thanks, I'll try to look into it. My ultimate goal is to have the same (or mostly same) code across various platforms I support.

closed 2024-03-18 by petzval