petzval/btferret #34: Support for BLE connection with 6-digit passkey authentication

jgocm, 2024-01-30:
Does your library provide support for BLE connection with 6-digit passkey authentication? 

If yes, could you please provide any examples on how to implement a BLE Client and connect it to a server using this functionality?

petzval, 2024-01-31:
I think the answer is probably no. A Classic connection does implement passkey authentication, so the code is there if the server requests a key, but I have not tested it with a BLE device that requires a secure encrypted connection. It can be quickly checked by running btferret which can be a BLE client. Run an LE scan (b command) to find the BLE device, then try and connect with the c command.

jgocm, 2024-01-31:
After a few tries I could connect and check the services from my BLE server, however they could not be read due to "Insufficient Authentication". Here are my command line inputs and outputs (after the scan):
```
> c

CLASSIC servers + LE servers + NODE servers - disconnected only
 2 - mcp2                        1010 - LE node 1010
 1000 - LE node 1000       1011 - LE node 1011
 1001 - LE node 1001       1012 - LE node 1012
 1002 - LE node 1002       1013 - LE node 1013
 1003 - LE node 1003       1014 - LE node 1014
 1004 - LE node 1004       1015 - LE node 1015
 1005 - Moura Lock Rack    1016 - LE node 1016
 1006 - LE node 1006       1017 - LE node 1017
 1007 - LE node 1007       1018 - LE node 1018
 1008 - LE node 1008       1019 - LE node 1019
 1009 - LE node 1009       1020 - LE node 1020
Input node  (x=cancel)
? 1005

CONNECTION COMPLETE TIME
  After connecting, some LE servers need more time to complete
  the process or they will disconnect. Zero may work, otherwise
  find the shortest time that will prevent disconnection.
  Current value=3000  (x=cancel to keep)  Default value=750
Time in ms  (x=cancel)
? 3000
Connect OK as LE client
> v

Read services

CLASSIC servers + LE servers + NODE servers
 1 - Local                 1010 - LE node 1010
 2 - mcp2                        1011 - LE node 1011
 1000 - LE node 1000       1012 - LE node 1012
 1001 - LE node 1001       1013 - LE node 1013
 1002 - LE node 1002       1014 - LE node 1014
 1003 - LE node 1003       1015 - LE node 1015
 1004 - LE node 1004       1016 - LE node 1016
 1005 - Moura Lock Rack    1017 - LE node 1017
 1006 - LE node 1006       1018 - LE node 1018
 1007 - LE node 1007       1019 - LE node 1019
 1008 - LE node 1008       1020 - LE node 1020
 1009 - LE node 1009
Input node  (x=cancel)
? 1005
Reading LE services from Moura Lock Rack.....
  Error 5 Insufficient Authentication
  Error reading characteristic index 4
  Error 2 Read Not Permitted
  Error reading characteristic index 5
  Error 5 Insufficient Authentication
  Error reading characteristic index 6
  Error 5 Insufficient Authentication
  Error reading characteristic index 7
  Error 5 Insufficient Authentication
  Error reading characteristic index 8
  Error 5 Insufficient Authentication
  Error reading characteristic index 9
Characteristics saved to device info
   ctic
   index       LE Characteristics
     0  Device Name          15 byte Permit 0A rwa Handle=0003 UUID=2A00
     1  Appearance           2 byte Permit 02 r   Handle=0005 UUID=2A01
     2  Pref Connection Parameters         8 byte Permit 02 r   Handle=0007 UUID=2A04
     3  Central Address Resolution         1 byte Permit 02 r   Handle=0009 UUID=2AA6
     4  4C0D1101F92E96A6C7412FC17F1BA1A0   ? byte Permit 12 rn  Handle=000D 
            UUID=4C0D1101F92E96A6C7412FC17F1BA1A0
     5  2E3714015432FC8A2746CF2B3ACD73DF   ? byte Permit 0A rwa Handle=0011 
            UUID=2E3714015432FC8A2746CF2B3ACD73DF
     6  131D1101F0611FEEC1F2FFAD67001110   ? byte Permit 1A rwan Handle=0014 
            UUID=131D1101F0611FEEC1F2FFAD67001110
     7  87881101FEB450A0874D17E81166E571   ? byte Permit 02 r   Handle=0018 
            UUID=87881101FEB450A0874D17E81166E571
     8  87881102FEB450A0874D17E81166E571   ? byte Permit 02 r   Handle=001A 
            UUID=87881102FEB450A0874D17E81166E571
     9  87881103FEB450A0874D17E81166E571   ? byte Permit 02 r   Handle=001C 
            UUID=87881103FEB450A0874D17E81166E571
>  
```

Any suggestions on how could we add this feature (passkey authentication for BLE) to the library?

petzval, 2024-01-31:
As I suspected, the BLE device wants a secure connection. Authentication for LE devices has not been implemented and would not be simple because there are multiple key exchanges. But there is one un-documented and un-tested thing you can try. Other than this, the only option would be to use bluez which has all the security features.

```
In btlib.c (version 13 - recently uploaded) at line 4345 in leconnect function
Insert lepair call as follows:

  ....
  popins();
  flushprint();
// pair here
  lepair(ndevice);
  return(1);
  } 
```

jgocm, 2024-01-31:
Unfortunately, I guess this idea was not sufficient to get it working. Here is my output:

```
> c

CLASSIC servers + LE servers + NODE servers - disconnected only
 1 - My Pi              1013 - LE node 1013
 1001 - LE node 1001    1014 - LE node 1014
 1002 - LE node 1002    1015 - Moura Lock Rack
 1003 - LE node 1003    1016 - LE node 1016
 1004 - LE node 1004    1017 - LE node 1017
 1005 - LE node 1005    1018 - LE node 1018
 1006 - LE node 1006    1019 - LE node 1019
 1007 - LE node 1007    1020 - LE node 1020
 1008 - LE node 1008    1021 - LE node 1021
 1009 - LE node 1009    1022 - LE node 1022
 1010 - LE node 1010    1023 - LE node 1023
 1011 - LE node 1011    1024 - LE node 1024
 1012 - LE node 1012    1025 - LE node 1025
Input node  (x=cancel)
? 1015

CONNECTION COMPLETE TIME
  After connecting, some LE servers need more time to complete
  the process or they will disconnect. Zero may work, otherwise
  find the shortest time that will prevent disconnection.
  Current value=750  (x=cancel to keep)  Default value=750
Time in ms  (x=cancel)
? 3000
Connect OK as LE client
PAIR FAIL
> 
```

petzval, 2024-02-01:
There may be a simple fix, or the BLE device may be requesting a higher security level than programmed in lepair. If you want one more try, it is necessary to see the Bluetooth traffic that is causing the pair failure. It can be sent to a file "btout.txt" with the following procedure.

```
In btferret

k command
  input option 2 - verbose

connect to LE device - Bluetooth traffic is displayed

o command
   saves screen output to file btout.txt
```

jgocm, 2024-02-02:
Yes, I have interest in that solution! In our case, we need to access the characteristics of a BLE server with a fixed passkey.

Thank you for the support and fast replies!

petzval, 2024-02-02:
Four modifications as follows
1.
```
AT LINE 4368 - replace lepair function
             - call lepair() as before at LINE 4345

int lepair(int ndevice)
  {  
  int n;
  preq[PAKHEADSIZE+10] = 0x04;  // kbd + dis
  preq[PAKHEADSIZE+12] = 0x04;  // auth mitm = passkey required
  sendhci(preq,ndevice); 
  readhci(ndevice,IN_AUTOEND,0,gpar.leclientwait,0);   // now user-set LE wait
  n = findhci(IN_AUTOEND,ndevice,INS_POP);
  if(n >= 0 && insdat[n] == AUTO_PAIROK)
    NPRINT "PAIR OK\n");
  else
    NPRINT "PAIR FAIL\n");
  popins();
  flushprint();
  return(1);
  }
```
2.
```
AT LINE 6715
REPLACE
  unsigned char key[16],out[16],ia[6],ra[6];
WITH
  unsigned char out[16],ia[6],ra[6];
  static unsigned char key[16];
```
3.
```
     AT LINE 7305 replace code block and code the passkey at j = ....
     
     if(insdat[n] == 2)
       {
       VPRINT "GOT pair response - SEND I confirm\n");
       // save for confirm calc
       for(j = 0 ; j < 7 ; ++j)
         sp->pair[j] = insdat[n+j];  // response 
                           
       for(j = 0 ; j < 16 ; ++j)
         sp->irand[j] = rand() & 0xFF;

       for(j = 0 ; j < 16 ; ++j)
         key[j] = 0;

       // IF passkey is fixed e.g. 123456
       j = 123456;
        
       // *** OR if the passkey is displayed on the remote device
       // printf("Input passkey\n? ");
       // j = setkeymode(0);
       // fgets(buf,16,stdin);
       // setkeymode(j); 
       // j = atoi(buf);
       // *** end input passkey

       key[0] = j & 0xFF;
       key[1] = (j >> 8) & 0xFF;
       key[2] = (j >> 16) & 0xFF;
       key[3] = (j >> 24) & 0xFF;

       calcc1(key,sp->irand,preq+PAKHEADSIZE+9,sp->pair,0,rp->leaddtype & 1,ia,ra,confirm+PAKHEADSIZE+10);
       sendhci(confirm,devicen);
       }
```
4.
```
     AT LINE 7332 - remove two instructions
   
     else if(insdat[n] == 4)
       {
       VPRINT "GOT R random\n");
       for(j = 0 ; j < 16 ; ++j)
         {
         sp->rrand[j] = insdat[n+j+1];
         // key[j] = 0;  ############### REMOVE THIS #########
         }
       calcc1(key,sp->rrand,preq+PAKHEADSIZE+9,sp->pair,0,rp->leaddtype & 1,ia,ra,out);
       getout = 0;
       for(j = 0 ; j < 16 && getout == 0 ; ++j)
         {
         if(sp->confirm[j] != out[j])
           getout = 1;
         }
       if(getout == 0)
         {
         VPRINT "R confirm OK\n");
         sp->cryptoflag = 1;
         }
       else
         VPRINT "I confirm fail\n");
         
       VPRINT "SEND LTK encrypt\n");
       calcs1(key,sp->rrand,sp->irand,sp->linkey);
       for(j = 0 ; j < 16 ; ++j)
         {
         // key[j] = 0;  ############ REMOVE THIS ############
         ltkcrypt[j+PAKHEADSIZE+16] = sp->linkey[j];
         }
       sendhci(ltkcrypt,devicen);
       }
```

jgocm, 2024-02-15:
Hi, I've only been able to test it today.

Using your modifications, it still returned me with PAIR FAIL. With a deeper analysis, I could find out that my `n` and `insdat[n]`, right before `if(n >= 0 && insdat[n] == AUTO_PAIROK)`, are :
> n:0, insdat[n]:5

I could not figure out what those variables mean just by looking through the code, but I've noticed that that `insdat[n]` should be equals to 4 to get a PAIR OK.

Would you have any guess on how we can fix that?

petzval, 2024-02-15:
One quick thing to try, just in case the stored key is at fault. Delete the link.key file. It will be in the directory from which the code was run.

jgocm, 2024-02-15:
I have no link.key files here. 

And, just like your implementation, my device has a fixed passkey=123456.

jgocm, 2024-02-19:
Got the following response:

> Connect OK as LE client
> GOT pair response - SEND I confirm
> GOT R confirm - SEND I rand
> GOT R random
> I confirm fail
> SEND LTK encrypt 00
> PAIR FAIL FD 40 08 00
> Moura Lock Rack has disconnected
> n:0, insdat[n]:5
> PAIR FAIL

petzval, 2024-02-19:
If your passkey is 123456 then the value of key[] is being corrupted. Make sure that it is declared as static: at the top of the immediate() function.

```
static unsigned char key[16];
```

jgocm, 2024-02-19:
We got it to work with this last change! Thank you for the support!

I've opened PR #35 to merge these changes into the original code.

Would you mind *deleting* this issue?

petzval, 2024-02-19:
OK, good. This has been very helpful in sorting out LE pairing, and it will be added as a new feature in the next release with an le_pair() function, so the changes will appear in the next few weeks.

The protocol is in the Core Specification document from here:

https://www.bluetooth.com/specifications/specs/core-specification/

Volume 3, Part H, Appendix C, Sections C1.1 and C2.1.2
Details of the HCI packets in Sections 3.5 and 3.6

jgocm, 2024-02-19:
Nice, thank you again! 

I've deleted some of my comments as I might have shared some "non-public" information unintentionally.

I am closing this issue.

closed 2024-02-19 by jgocm