petzval/btferret #36: Reference miss match

ashoksharma842, 2024-03-21:
In Readme file,

> 6 Documentation References

There is no **Part E** in **Volume 4** of _BLUETOOTH SPECIFICATION Version 5.0_

petzval, 2024-03-21:
The specification document has been changed since version 5.0. The latest version is 5.4, which has a Part E.

ashoksharma842, 2024-03-22:
Now it is in **volume 2 HOST CONTROLLER INTERFACE FUNCTIONAL SPECIFICATION**, 

- Part E : HOST CONTROLLER INTERFACE FUNCTIONAL SPECIFICATION

7. HCI COMMANDS AND EVENTS
7.1 LINK CONTROL COMMANDS
7.2 LINK POLICY COMMANDS
7.3 CONTROLLER & BASEBAND COMMANDS
7.4 INFORMATIONAL PARAMETERS
7.5 STATUS PARAMETERS
7.6 TESTING COMMANDS
7.7 EVENTS
7.8 LE CONTROLLER COMMANDS

petzval, 2024-03-22:
It was in volume 2 before it was moved to volume 4, and I had to re-write all the references. This is the latest version:

https://www.bluetooth.com/specifications/specs/core-specification-5-4/

closed 2024-04-01 by ashoksharma842