petzval/btferret #37: I can't see the esp-32 device name in BLE SCAN.

joseeduardolins, 2024-03-21:
I programmed an ESP32 to be a BLE server using the BLEDevice.h library, in arduino IDE. When using the btferret code, I can scan the ESP32's signal using its MAC address. However, the device's name that I configured in the ESP32's firmware code does not appear in the btferret scan. Interestingly, I can read the device's name if I perform a characteristic read. Has anyone encountered a similar issue or know how to resolve it?

This is a ESP32 firmware code:

```
    #include <BLEDevice.h>
    #include <BLEUtils.h>
    #include <BLEServer.h>
    #include "heltec.h"

    #define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
    #define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"
    #define DEVICE_NAME "ESP 32 LORA"

    #define servGen "0x1800"
    #define carac "0x2A00"

    void setup() {
    Heltec.begin(true /*DisplayEnable Enable*/, false /*LoRa Disable*/, true /*Serial Enable*/);
    Heltec.display->clear();

    Heltec.display->setFont(ArialMT_Plain_10);
    Heltec.display->setTextAlignment(TEXT_ALIGN_LEFT);
    Heltec.display->drawStringMaxWidth(0, 0,128,"Device name:" + String(DEVICE_NAME));
    Heltec.display->drawStringMaxWidth(0,25,128, "Characteristic UUID: " + String(CHARACTERISTIC_UUID) );

    Heltec.display->display();

     Serial.println("Starting BLE work!");

     BLEDevice::init(DEVICE_NAME);
     BLEServer *pServer = BLEDevice::createServer();
     BLEService *pService = pServer->createService(SERVICE_UUID);
     BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

     pCharacteristic->setValue("Caract ESP32");
     pService->start();
     // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
     BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
     pAdvertising->addServiceUUID(SERVICE_UUID);
     pAdvertising->setScanResponse(true);
     BLEDevice::startAdvertising();
     Serial.println("Characteristic defined! Now you can read it in your phone!");
     }

     void loop() {
     // put your main code here, to run repeatedly:
    delay(2000);
     }
```
And this is the output after to LE scan (esp32 escaned):
```
   3 FOUND 3C:61:05:2B:BF:92 - Fixed
    Flags = 06
    All 16-byte UUIDs = 4B 91 31 C3 C9 C5 CC 8F 9E 45 B5 1F 01 C2 AF 4F
    Slave Connect Interval Range = 20 00 40 00
    Known device 1001 LE node 1001`
```
And this is the output after to read the esp32 services (the device name can be readed)
```
Reading LE services from LE node 1003.....
Characteristics saved to device info
   ctic
   index       LE Characteristics
     0  Service Changed      ? byte Permit 20 i   Handle=0003 UUID=2A05
     1  Device Name          11 byte Permit 02 r   Handle=0016 UUID=2A00
     2  Appearance           2 byte Permit 02 r   Handle=0018 UUID=2A01
     3  2AA6 UUID            1 byte Permit 02 r   Handle=001A UUID=2AA6
     4  BEB5483E36E14688B7F5EA07361B26A8   12 byte Permit 0A rwa Handle=002A
            UUID=BEB5483E36E14688B7F5EA07361B26A8
 ```
I appreciate any help, thanks.

petzval, 2024-03-21:
The device name can appear in two places:
1. Advertising data - can be seen by a scan
2. A characteristic with a UUID of 2A00 - not visible to a scan, only available by reading characteristic.

You have to add the name to the advertising data. I'm not familiar with ESP32 code, but looking at the example I would guess that it is done by something like:

```
    pAdvertising->addName("Device name");
```

joseeduardolins, 2024-03-23:
Thanks. In fact, the device name does not appear in the advertising. The device name is only stored in the UUID characteristic 2A00. I will correct this.

closed 2024-03-23 by joseeduardolins