petzval/btferret #38: error with apple iphone client and le_server on a raspberry pi 4 B with Bookworm OS

dghockey, 2024-03-23:
First, thanks for the tremendously great project you have put together with btferret! 
I'm working on a raspberry pi robot device that I want to control with a simple Blue Tooth app on an iphone. I use this iphone app effectively when controlling some ESP32 devices associated with this project, so it seems to work. However, when running server code on my RPi (for example, hello world example, or the le_server.py example, I get an error - 
PAIR fail - No bond info
The iphone connects (the server call back prints out the "connection" message), but then the error appears on the RPI terminal screen, and the connection fails. 

Here is my devices.txt file:
     
DEVICE = Desk Pi         TYPE=MESH    NODE=1  ADDRESS = D8:3A:DD:1D:1B:40
  PRIMARY_SERVICE = 1800
    LECHAR = Device name  PERMIT=06  SIZE=16  UUID=2A00   ; index 0
  PRIMARY_SERVICE = 112233445566778899AABBCCDDEEFF00
    LECHAR = My data  PERMIT=06  SIZE=16 UUID=ABCD        ; index 1    
    
DEVICE = ESP32 talkieBot     TYPE=LE      NODE=1002   ADDRESS = A0:B7:65:60:87:92

The server code I'm using is pretty much what you have shown as examples, with a few additional print messages to try and see what is going on.

What might I be doing incorrectly to get the no bond error message?
Thanks so much for any advice you might be able to provide
Dallas

petzval, 2024-03-24:
I do not have an iPhone, so I cannot check this myself. Here's how the process should work:

1. First connection. Client asks to bond. A new link key is generated by the server, and both devices store it.
2. Subsequent connections. Client has the stored link key, so it asks to re-pair using the stored keys instead
of going through the key generation again. The server gets the key from its store. If it does not have a key stored, it generates the "No bond info" error.

Has the first connection failed with the error? If so, the problem is that the client has a key stored from some previous non-btferret connection that it is trying to use, and btferret has no knowledge of it. See "Problems caused by connecting device" in section 3.11 of the documentation. It explains why the client's cache can cause this sort of trouble.  Can you wipe the client's cache? Or at least make sure the Pi is not paired to force a new key generation.

dghockey, 2024-03-24:
OK. Thanks for getting back to me.

I still can’t get le_server working with the iphone. I’ve decided to take a different approach that doesn’t require the iPhone. Thanks so much for your assistance. I’m sorry I couldn’t get it to work, but I gotta move on with my project.

Dallas

petzval, 2024-03-25:
OK, but this has been helpful. Because it seems that this might be a generic problem, I have uploaded a new le_server.py with the code for what should be a fix (but with the instructions commented out). Uncommenting the indicated instructions will set up a random address, which will create a new identity for the Pi that the client has not seen before.

dghockey, 2024-03-25:
Hey bud –

That fixed it! 

I can write commands and read responses to and from the RPi with my iPhone. 

I’m testing the Blue tooth interface on my desktop Pi. My robot pi actually runs a different RPi OS

(Buster). But not anticipating any issues since the problem was how the iPhone managed it’s connections.

Thanks so much.

Dallas

petzval, 2024-03-26:
OK, thanks. That's very useful information. I'll add something to the documentation.

dghockey, 2024-03-26:
Thanks again for your patience and assistance.

One last question ….

Does the btferret library have to be run only by users with superuser permissions? Or can the default user also run the code, perhaps by changing the lib file ownership and permission attributes?

 

Dallas

petzval, 2024-03-27:
Btferret needs root permissions for low-level access to the Bluetooth adapter.

dghockey, 2024-03-27:
Ok. Not a problem. Thanks