petzval/btferret #39: Minimum Debian/Ubuntu support

fromage9747, 2024-03-26:
I came across your library and have successfully run it in Debian 12.  I have a device that I can't upgrade and is on Debian 7 but as this uses the low level HCI commands, I thought that I could get it to run there.

I managed to compile it and start it, but it crashes my bluetooth and displays the message:

```
Initialising...
No root permission or Bluetooth (hci0) is off or crashed
Must run with root permission via sudo as follows:
  sudo ./btferret (for C) or  sudo python3 btferret.py
```

What would be the minimum version/OS that this can run on? Or maybe I need to update something?

Any help would be greatly appreciated.

fromage9747, 2024-03-26:
syslog output:

```
Mar 26 03:08:06 debian7 bluetoothd[3531]: HCI dev 0 unregistered
Mar 26 03:08:06 debian7 bluetoothd[3531]: Stopping hci0 event socket
Mar 26 03:08:06 debian7 bluetoothd[3531]: Unregister path: /org/bluez/3531/hci0
Mar 26 03:08:12 debian7 bluetoothd[3531]: HCI dev 0 registered
Mar 26 03:08:12 debian7 bluetoothd[3531]: Listening for HCI events on hci0
Mar 26 03:08:12 debian7 bluetoothd[3531]: HCI dev 0 up
Mar 26 03:08:12 debian7 bluetoothd[3531]: Proximity GATT Reporter Driver: Operation not permitted (1)
Mar 26 03:08:12 debian7 bluetoothd[3531]: sap-dummy interface org.bluez.SimAccessTest init failed on path /org/bluez/test
Mar 26 03:08:12 debian7 bluetoothd[3531]: Sap driver initialization failed.
Mar 26 03:08:12 debian7 bluetoothd[3531]: sap-server: Operation not permitted (1)
Mar 26 03:08:12 debian7 bluetoothd[3531]: Unable to load keys to adapter_ops: Function not implemented (38)
Mar 26 03:08:12 debian7 bluetoothd[3531]: Adapter /org/bluez/3531/hci0 has been enabled
Mar 26 03:16:40 debian7 bluetoothd[3531]: HCI dev 0 down
Mar 26 03:16:40 debian7 bluetoothd[3531]: Adapter /org/bluez/3531/hci0 has been disabled
```
Looks like this is the problem, `Unable to load keys to adapter_ops: Function not implemented (38)`

petzval, 2024-03-26:
I do not have Debian 7, so I can't check it myself. From the syslog it looks like hci0 is OK. Check by running:

```
hciconfig
```
See if you get a listing for hci0.
Try enabling or disabling hci0 first:

```
sudo hciconfig hci0 down
OR
sudo hciconfig hci0 up
```

Or try shutting down bluetooth before running btferret:

```
sudo service bluetooth stop
```

Or - this is a bit more complicated because the library must be edited. Get more information about the failure by enabling verbose mode and save the results to a text file as follows. I'm not sure if you're running C or Python. If Python - run btfpy.py build again.

```
In btlib.c file Line 1342

  set_print_flag(PRINT_NORMAL);  
  // set_print_flag(PRINT_VERBOSE);

CHANGE TO

  // set_print_flag(PRINT_NORMAL);  
   set_print_flag(PRINT_VERBOSE);

```
Then run this code.

```
C

int main()
  {
  init_blue("devices.txt");
  output_file("bug.txt");
  return(0);
  }

PYTHON

btfpy.Init_blue("devices.txt")
btfpy.Output_file("bug.txt")

```