petzval/btferret #41: Unable to connect BLE Server again after disconnecting.

ashoksharma842, 2024-04-29:
In BLE server, callback,

```
else if(operation == LE_DISCONNECT) {
    // clientnode has just disconnected
    // uncomment next line to stop LE server when client disconnects
    // return(SERVER_EXIT);
    // otherwise LE server will continue and wait for another connection
    // or operations from other clients that are still connected
    printf("Disconnected\n");
    // return(SERVER_EXIT);
}
```

on commenting return(SERVER_EXIT), the server shows disconnected 
but it doesn't connect again.
To connect again, binary needs restart.

petzval, 2024-05-01:
I have not been able to reproduce your error. If I connect to the server, then disconnect, the server goes back into waiting mode and I can re-connect with no problem.  Can you supply more details, it might be a problem with the client.

JonathanCortez, 2025-03-08:
I get this similar effect when I disconnect from my app or device, it shutdown the LE Server. But what works for me is by just removing return(SERVER_EXIT) this keeps it going and I can connect again.