petzval/btferret #42: runing le server gives: "Invalid local characteristic index in write_ctic"

pierre-gardyn, 2024-06-04:
First, thanks for this amazing library and the effort you put into that.

I compile and run the le server code you have provided the code for at:
https://github.com/petzval/btferret?tab=readme-ov-file#3-7-le-server

but when I run it I have the error:
Invalid local characteristic index in write_ctic

I also have this error on the timer code every 10 seconds when it tries to execute:
    buf[0] = 0x67;
    write_ctic(localnode(),6,buf,0);  

The error is triggered emitted by this line:
https://github.com/petzval/btferret/blob/d5f6d82122d7e934a2455f9fa380d2b9d17208c6/btlib.c#L5171

where the value of cp->type is  2 (CTIC_END).

The device data displayed when the application starts seems to indicate that the slot 6 is a variable:

DEVICE = My Pi  TYPE=Mesh  NODE=1   ADDRESS = 00:1E:C0:2D:17:7C
PRIMARY_SERVICE = 1800
  LECHAR = Device Name  PERMIT=02 SIZE=16 UUID=2A00
  LECHAR = Appearance   PERMIT=02 SIZE=2  UUID=2A01
PRIMARY_SERVICE = 1801
  LECHAR = Service changed PERMIT=20 SIZE=4 UUID=2A05
PRIMARY_SERVICE = 180A
  LECHAR = PnP ID          PERMIT=02 SIZE=7 UUID=2A50
PRIMARY_SERVICE = 112233445566778899AABBCCDDEEFF00
  LECHAR = Control  PERMIT=06 SIZE=2 UUID=ABCD
  LECHAR = Info     PERMIT=06 SIZE=1 UUID=CDEF
  LECHAR = Data     PERMIT=12 SIZE=1 UUID=DEAF

petzval, 2024-06-05:
One thing to check first. Have you set the local board ADDRESS= in devices.txt? If the local address has been set, then the value of localnode() will be the NODE=number which is 1. If not, localnode() will be 1000 or higher.

pierre-gardyn, 2024-06-06:
You were right. I added those lines (that you have in mouse.c) to confirm:

```
  if(localnode() != 1)
    {
    printf("ERROR - Edit mouse.txt to set ADDRESS = %s\n",device_address(localnode()));
    return(0);
    }
```

You cn close that issue. Thanks again

closed 2024-06-08 by petzval