petzval/btferret #45: Save bonding info to file after receiving it

anujdeshpande, 2024-08-10:
Hi

I am trying to figure how to save the bonding info to the `/etc/btferret.dat` file right after pairing has succeeded. Right now, when I start the server from `python3 btferret.py` and then `s` and `LE server` - the pairing information gets saved to the file in `etc` when I close the server.
The problem with that kind of an approach is that if my application closes unexpectedly, I might lose the information in memory.

petzval, 2024-08-11:
The btferret.dat file is written on program termination by this call in close_all in btlib.c:
```
at LINE 5115
rwlinkey(1,0,NULL);
```
There is some protection against close_all not being called explicitly by using atexit which specifies functions that are called when the program exits:
```
at LINE 2017
atexit(close_all);
```
This should work when the program terminates, but if not, to save the data at any other time, call rwlinkey(1,0,NULL). In C, this is pretty simple, just declare rwlinkey in btlib.h or in your code. It will then be callable from your C code.
```
void rwlinkey(int rwflag,int ndevice,unsigned char *addr); 
```
In Python it is more complicated because function calls go via btfpython.c, and rwlinkey would need its own entry. An easier option is to piggy-back on an existing Python call. I would suggest Set_flags: 
```
at LINE 13062 in btlib.c modify as follows

void set_flags(int flags,int onoff)
  {
  if(flags == 0)
    {
    rwlinkey(1,0,NULL);
    return;
    }
  if(onoff == FLAG_OFF)
    gpar.settings &= ~flags;
  else
    gpar.settings |= flags;
  }  

Then recompile the Python module with "python3 btfpy.py build".

Call from Python code via:

  btfpy.Set_flags(0,0)
```  
I don't think there will be any problems with this, but I cannot be absolutely sure.

anujdeshpande, 2024-08-11:
@petzval thanks for the detailed info - I will try this out and report here.

Are you open to accepting a PR where I add this as a separate API? I'd rather not maintain a fork if I can avoid it.

petzval, 2024-08-12:
I've had a careful look at rwlinkey and there is a problem with calling it before program termination. If the device disconnects, a re-connection during the same session will probably fail. Some minor changes will fix this, and a revised version of rwlinkey is below. Rather than maintaining a PR, I would rather just add a new function to save the pairing info and do a full test to make sure it works . I can also see the value of adding an empty user-defined function inside btlib.c. These will be in a new version within a week or so.

```
void rwlinkey(int rwflag,int ndevice,unsigned char *addr)
  {
  int n,k,i,j,addcount,flag;
  unsigned char *badd,*key;
  struct devdata *dp;
  FILE *stream;
  static char *fname = "/etc/btferret.dat";
  static int count = -1;
  static int delflag = 0;
  static int writeflag = 0;
  static unsigned char zero[6] = {0,0,0,0,0,0};
  static unsigned char *table = NULL; 
  
        
  if(rwflag == 0)
    {
    // read
    if(ndevice > 0 && (dev[ndevice]->linkflag & (KEY_NEW | KEY_FILE)) != 0)
      return;   // dev[]->linkey is good
      
    if(count < 0)
      {
      count = 0;  // in file
      stream = fopen(fname,"rb");
      if(stream == NULL)
        return;
 
      n = 0;
      count = fgetc(stream);
      
      if(count != 0 && count != 0xFF)
        {
        k = count*22;
        table = (unsigned char *)malloc(k);

        if(table != NULL && fread(table,1,k,stream) == (unsigned int)k)
          n = 1;
        }
             
      fclose(stream);
      if(n == 0)
        {
        count = 0;
        NPRINT "Read key data failed\n");
        return;
        }
      }
     
    for(k = 0 ; k < count && table != NULL ; ++k)
      {
      badd = table + k*22;
      key = badd+6;
      if(addr == NULL)
        n = devnfrombadd(badd,BTYPE_CL | BTYPE_LE | BTYPE_ME,DIRN_FOR);  
      else
        {
        n = bincmp(badd,addr,6,DIRN_FOR);
        if(n != 0)
          n = ndevice;
        }  
      if( (ndevice == 0 && n > 0) || (ndevice > 0 && n == ndevice) )
        {  // all on init (ndevice=0)  or ndevice only
        dp = dev[n];
        if((dp->linkflag & KEY_FILE) != 0 && bincmp(zero,key+10,6,DIRN_FOR) != 0)
          {
          for(n = 0 ; n < 10 ; ++n)
            dp->divrand[n] = key[n];
[... 132 lines cut ...]
    } 
  else if(rwflag == 3 && count > 0) 
    {
    for(k = 0 ; k < count && table != NULL ; ++k)
      {
      badd = table + k*22;
      n = devnfrombadd(badd,BTYPE_CL | BTYPE_LE | BTYPE_ME,DIRN_FOR);
      if(n >= 0)
        NPRINT "%s (%02X) =",dev[n]->name,dev[n]->linkflag >> 10);
      else
        NPRINT "Unknown ="); 
      
      for(j = 0 ; j < 22 ; ++j)
        {
        NPRINT " %02X",badd[j]);
        if(j == 5)
          NPRINT " =");
        }
      NPRINT "\n");
      flushprint();
      }
    }
  }  

```

petzval, 2024-08-13:
I can now see other problems associated with calling rwlinkey twice, so I cannot recommend the changes I have suggested. A new version will have a fix.

anujdeshpande, 2024-08-19:
Thanks a lot @petzval - will try it out and get back!

anujdeshpande, 2024-08-20:
@petzval do you think it makes sense to have an event for that? Maybe `LE_BONDED` - so that user can take further action in the callback