petzval/btferret #50: Running BLE and classic at the same time

gregoiregentil, 2024-09-10:
Following issue 48 (everything is working fine on BLE both client and server at the same time), I'm still interested to run a BLE LE_server and a Bluetooth Classic server at the same time on the same device. Could you please explain from where comes the limitation? Is this "patch-able"? Would the limitation comes from the implementation of the btferret library and/or could the bluetooth chip prevents from doing both protocols at the same time (and how would I test that)?

petzval, 2024-09-10:
I think it would be possible, but it would require a major reorganization to merge the btilb.c functions le_server and classic_server. They both have a section that waits for a connection, and a do-loop that waits for packets from the connected client and deals with them. In classic_server the connection and input sections are separate which means that only one connection is possible. In le_server the connection section is in the do loop, so multiple connections are allowed. The merged procedure would have a do-loop that waits for classic and le connections, and also waits for connected classic and le packets, tests each input for classic or le and switches the handling accordingly.

gregoiregentil, 2024-09-11:
Thank you for the clear explanation.

I think that's too bad because you are very close to support such scenario.

This is where having a poll architecture WITHOUT blocking as discussed in issue #23 is highly beneficial. And it's not so much to implement, even from your existing code. You have a main poll and you dispatch depending whatever needs to be done. And the advantage of non blocking is that you are still getting events (they are queued by the kernels) and you can deal with them.

I'm closing the issue as it's more an enhancement that a bug.

closed 2024-09-11 by gregoiregentil