petzval/btferret #52: Is it possible to call notify_ctic without discovering find_ctics as a BLE client?

gregoiregentil, 2024-09-10:
As a btferret BLE client, is there a way to call notify_ctic(NODE_SERVER, uiid, NOTIFY_ENABLE, notify_callback) without discovering the characteristics with find_ctics(NODE_SERVER) ? Obviously, uiid is known on the client side.

If find_ctics is not called, btlib returns "Invalid node or characteristic index in write_ctic".

I have tried to put the known characteristics below the NODE_SERVER in the configuration file without any success.

petzval, 2024-09-10:
You can avoid calling find_ctics by putting all the required information in the devices.txt file. Most important is the HANDLE setting. From btferret use v - Read services (which is find_ctics) and then i - Device information to display the handle values, then put the handle values in the devices.txt LECHAR entries.

gregoiregentil, 2024-09-10:
Working. Thank you.

closed 2024-09-10 by gregoiregentil