petzval/btferret #53: How to combine mouse and keyboad?

futurestandard, 2024-09-19:
I am currently experimenting with using a Raspberry Pi 4 as a PC mouse and keyboard. The mouse and keyboard examples work individually, but since they occupy the device as root, I am unable to use both at the same time.

In the case of the ESP32 sample (https://github.com/blackketter/ESP32-BLE-Combo/tree/master), there is a combined implementation, so I believe it's technically possible, and I've been trying to achieve the same.

However, after merging the report map and report for both keyboard and mouse, and adjusting the size in the text file, the device disconnects with an error a few seconds after pairing. Since I can't determine the exact details of the device error, could you provide guidance on what aspects I should focus on?

I plan to use this for direct control after screen reading with AI generation.

petzval, 2024-09-21:
The problem will be in the way the report maps are combined. I will try it and report back - it may take a week.

petzval, 2024-09-23:
1. From mouse.c reportmap[52] change [7] from 01 to 02. This is the mouse report ID
2. Merge reportmap[47] from keyboard.c and modified reportmap[52] from mouse.c into a single reportmap[99] = 05,01,09,06.......81,06,C0,C0
3. Modify keyboard.txt
''''
LECHAR=Report Map SIZE=99....change size
LECHAR=Report1   SIZE=8...no change
LECHAR=Report2   SIZE=3...add from mouse.txt
'''
4. In keyboard.c + mouse.c combined code send_key function use reportindex to send keystrokes and reportindex+1to send mouse commands

toriumi, 2024-10-05:
Thanks for your support.

Actually I have already apply those (for python). driver error has occurd.

I understood the reportmap, but I have no idea for report
keyboard send 8byte and mouse send 3byte, So I confuse report = [0,0,0,0,0,0,0,0]  or  [0,0,0,0,0,0,0,0,0,0,0]
and
uuid = [0x2A,0x4D]
btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid),report,0)
is correct or 

uuid = [0x2A,0x4D]
btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid),report1,0)
btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid),report2,0)

I attach the code here

hid_combined.txt
`DEVICE = My Pi   TYPE=Mesh  node=1  ADDRESS = DC:A6:32:70:DA:16
  PRIMARY_SERVICE = 1800
    LECHAR=Device Name   SIZE=4   Permit=02 UUID=2A00  
    LECHAR=Appearance    SIZE=2   Permit=02 UUID=2A01  
  PRIMARY_SERVICE = 180A
    LECHAR= PnP ID           SIZE=7 Permit=02   UUID=2A50  
  PRIMARY_SERVICE = 1812
    LECHAR=Protocol Mode   SIZE=1  Permit=06  UUID=2A4E  
    LECHAR=HID Info        SIZE=4  Permit=02  UUID=2A4A  
    LECHAR=HID Ctl Point   SIZE=8  Permit=04  UUID=2A4C   
    LECHAR=Report Map      SIZE=99 Permit=02  UUID=2A4B  
    LECHAR=Report1         SIZE=8  Permit=92  UUID=2A4D  
        ; Report1 must have Report ID = 1 
        ;   0x85, 0x01 in Report Map
    LECHAR=Report2         SIZE=3  Permit=92  UUID=2A4D  
        ; Report2 must have Report ID = 2 
        ;   0x85, 0x02 in Report Map
        ; unsigned char uuid[2]={0x2A,0x4D};
        ; index = find_ctic_index(localnode(),UUID_2,uuid)+1;
        ; Send data: write_ctic(localnode(),index,data,0);

;  *** Optional battery level ***
;  PRIMARY_SERVICE = 180F
;    LECHAR=Battery Level   SIZE=1 Permit=12  UUID=2A19   `

hid_combined.py
`#!/usr/bin/python3
import btfpy

# ********** Bluetooth keyboard **********
# From https://github.com/petzval/btferret
#   Build btfpy.so module - instructions in README file
#
# Download
#   keyboard.py
#   keyboard.txt
# 
# Edit keyboard.txt to set ADDRESS=
# to the address of the local device
# that runs this code
#
# Run
#   sudo python3 keyboard.py
#
# Connect from phone/tablet/PC to "HID" device
#
# All keystrokes go to connecting device
# F10 sends "Hello" plus Enter
# ESC stops the server 
#
# To add a battery level service:
# uncomment all battery level labelled
[... 294 lines cut ...]
  # btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid),battery,1) 
  #************************     
                          
  # Set unchanging random address by hard-coding a fixed value.
  # If connection produces an "Attempting Classic connection"
  # error then choose a different address.
  # If set_le_random_address() is not called, the system will set a
  # new and different random address every time this code is run.  
 
  # Choose the following 6 numbers
  # 2 hi bits of first number must be 1
randadd = [0xD3,0x56,0xDB,0x15,0x32,0xA0]
btfpy.Set_le_random_address(randadd)
     
btfpy.Keys_to_callback(btfpy.KEY_ON,0)   # enable LE_KEYPRESS calls in lecallback
                                         # 0 = GB keyboard  
btfpy.Set_le_wait(20000)  # Allow 20 seconds for connection to complete
                                         
btfpy.Le_pair(btfpy.Localnode(),btfpy.JUST_WORKS,0)  # Easiest option, but if client requires
                                                     # passkey security - remove this command  

btfpy.Le_server(lecallback,0)
  
btfpy.Close_all()
`

toriumi, 2024-10-05:
Thanks for your support.

Actually I have already apply those (for python). driver error has occurd.

I understood the reportmap, but I have no idea for report
keyboard send 8byte and mouse send 3byte, So I confuse report = [0,0,0,0,0,0,0,0]  or  [0,0,0,0,0,0,0,0,0,0,0]
and
uuid = [0x2A,0x4D]
btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid),report,0)
is correct or 

uuid = [0x2A,0x4D]
btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid),report1,0)
btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid),report2,0)

I attach the code here
[hid_combined.txt](https://github.com/user-attachments/files/17266795/hid_combined.txt)
[hid_combined.py.txt](https://github.com/user-attachments/files/17266796/hid_combined.py.txt)

petzval, 2024-10-06:
This should work - note +1 in report2 write index.

```
report1 = [0,0,0,0,0,0,0,0]
report2 = [0,0,0]
uuid = [0x2A,0x4D]
btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid),report1,0)
btfpy.Write_ctic(node,btfpy.Find_ctic_index(node,btfpy.UUID_2,uuid)+1,report2,0)
```

petzval, 2024-10-06:
Also, your report map needs a modification to set the ID of Report 2 to 02  (item [54] = 02 as below)

```
reportmap = [0x05,0x01,0x09,0x06,0xA1,0x01,0x85,0x01,0x05,0x07,0x19,0xE0,0x29,0xE7,0x15,0x00,\
             0x25,0x01,0x75,0x01,0x95,0x08,0x81,0x02,0x95,0x01,0x75,0x08,0x81,0x01,0x95,0x06,\
             0x75,0x08,0x15,0x00,0x25,0x65,0x05,0x07,0x19,0x00,0x29,0x65,0x81,0x00,0xC0,\
             0x05,0x01,0x09,0x02,0xA1,0x01,0x85,0x02,0x09,0x01,0xA1,0x00,0x05,0x09,0x19,0x01,\
             0x29,0x03,0x15,0x00,0x25,0x01,0x95,0x03,0x75,0x01,0x81,0x02,0x95,0x01,0x75,0x05,\
             0x81,0x01,0x05,0x01,0x09,0x30,0x09,0x31,0x15,0x81,0x25,0x7F,0x75,0x08,0x95,0x02,\
             0x81,0x06,0xC0,0xC0]
```