petzval/btferret #54: Error 3 Write Not Permitted

tkrmnz, 2024-10-09:
Revisiting the write notify project from my previous issue request.  I'm getting an "Error 3 Write Not Permitted" error.  Same as before, the notify can be triggered via CYSmart app on windows and Android BLE serial terminal.  I'm running V18 of btlib.c.

**Here is the devices.txt**
  DEVICE=BLE device TYPE=LE NODE=2  ADDRESS=60:B6:47:62:59:8A
  LECHAR=Data HANDLE=001D PERMIT=14 SIZE=60 UUID=FEC26EC46D7144429F8155BC21D658D6  ; index 0  
  LECHAR=CCC1 HANDLE=001E  PERMIT=08 SIZE=2 UUID=2902  ; index 1  

**Here is the LE_Client.c**
#include <stdio.h>
#include <stdlib.h>
#include "btlib.h"

int notify_callback(int lenode,int cticn,unsigned char *buf,int nread);
int main()
  {  

  unsigned char data[8];
  data[0] = 0x01;
  data[1] = 0x00;  
  if(init_blue("devices.txt") == 0)
    return(0);

  connect_node(2,CHANNEL_LE,0);
  
  //write_ctic(2,1,data,0);
  notify_ctic(2,0,NOTIFY_ENABLE,notify_callback);
  
    // read notifications for 1 minute
  read_notify(10000); 
  output_file("monitor2.txt");

  disconnect_node(30);
  close_all();
  return(0);
  } 

int notify_callback(int lenode,int cticn,unsigned char *buf,int nread)
  { 
  // LE device lenode has sent notification of characteristic index cticn
  // data in buf[0] to buf[nread-1]

  printf("Notify %02X %02X %02X\n",buf[0],buf[1],buf[2]);

  return(0);
  }  

**Monitor.txt output**

Device data from devices.txt file
DEVICE=BLE device TYPE=LE NODE=2  ADDRESS=60:B6:47:62:59:8A
  LECHAR=Data HANDLE=001D PERMIT=14 SIZE=60 UUID=FEC26EC46D7144429F8155BC21D658D6
  LECHAR=CCC1 HANDLE=001E  PERMIT=08 SIZE=2 UUID=2902

This local device has been allocated NODE = 1000
It should be added to the devices.txt file as follows:
DEVICE=name (e.g. My Pi) TYPE=MESH  NODE=choose (e.g. 1)  ADDRESS=DC:A6:32:0C:FC:BC
Connect OK as LE client
  Error 3 Write Not Permitted

petzval, 2024-10-09:
Error 3 is returned by the Bluetooth adapter, it is not a btlib-generated error. Now that btlib has been fixed to send a write with acknowledge read, it is not necessary to include the notify UUID 2902 in devices.txt. So first thing to try is delete the LECHAR=CCC1 entry.

tkrmnz, 2024-10-09:
I found the issue, HANDLE went from 001D to 001B.  I also removed the CCC1 entry in the Devices.txt.   Again, thank you for your assistance.

closed 2024-10-09 by tkrmnz