petzval/btferret #56: Keyboard release latency after long press

loOong-Cheng, 2024-12-03:
Hi,

Thanks for the amazing repository. I have tested the keyboard.c and keyboard.py using rpi4 and found that when I keep pressing one button and then release it, the android/PC would keep adding characters for quite some time. Is this caused by the throughput? 

Best

petzval, 2024-12-03:
If you are holding the button down for some time, then you are seeing the normal key repeat that is generated by the operating system, and the keyboard code is correctly receiving multiple LE_KEYPRESS calls. If the key press is short, but multiple characters appear on the client, then it a a tricky problem caused by the Bluetooth adapter. The send_key function sends the key code to signal a key down, and then immediately a "no key pressed" all-zero packet. The Bluetooth adapter can only cope with packets sent at a certain rate, and if they are sent too fast, then packets are lost. So if the no key pressed packet is lost, the client thinks the key is held down. The packet transmissions are deliberately slowed down to avoid this, but may need to be even slower - so in send_key add a delay between the key down write_ctic packet and the all-zero key up packet.

petzval, 2024-12-06:
It sounds like the opposite problem - too much delay. Remove the 0.04 delay and run this from the command line before starting keyboard. It slows the keyboard repeat rate.

```
kbdrate -r 2.0 -d 1000
```

closed 2024-12-11 by loOong-Cheng