petzval/btferret #58: Using a real mouse in mouse.py

Hitan999, 2024-12-17:
Hello, thank you very much for developing this. It works amazingly!
I am currently using it as a HID for keyboard, but I was wondering if it is possibile to use a "real" mouse (not the keyboard keys) for the report map. Thank you!

petzval, 2024-12-17:
Add a timer call to the le_server, and read mouse data in the LE_TIMER operation as below. I think that mouse0 just sends three bytes on every mouse movement  (button,x,y) and it might be as simple as sending those same three bytes to send_key(x,y,button). I don't think there is a problem with waiting in LE_TIMER for mouse input. If the response is slow it may be necessary to speed up the le server loop - see the closed isssue #55. 
```
le_server(lecallback,1);  // LE_TIMER every 100ms

//  in lecallback....

if(op == LE_TIMER)
  {
  // read /dev/input/mouse0
  // send data to send_key()
  }

Hitan999, 2024-12-19:
Thank you for your answer. I managed to read the mouse data and pass it to send_key(x,y,but), but currently is very delayed. If I move the mouse for one second, it will take two seconds to move the pointer.

I would like to try speeding up the server loop. I tried looking at line #55 but honestly I did not understand where is the line to increase frequency. Could you please help?

petzval, 2024-12-20:
This works for me with no changes to the server loop. On my Windows machine, the Y direction is opposite sign, so the Y byte sign has to be reversed. It seems that with a timer setting of 1, the key input routines do not have enough time to work and key inputs, including ESC to stop the server do not work - I'll look at this as a possible bug.

```
   // ADD these includes
#include <unistd.h>
#include <fcntl.h>

  ......
  
   // Set timerds = 1 
  le_server(lecallback,1);
  
  close_all();
  return(1);
  }

int lecallback(int clientnode,int op,int cticn)
  {
  int nread;
  char buf[3];
  static int fd = 0;       
         
  if(op == LE_CONNECT)
    {
    printf("Connected OK\n");
    fd = open("/dev/input/mouse0",O_NONBLOCK);
    }
  if(fd > 0 && op == LE_TIMER)
    {
    nread = read(fd,buf,3);
    if(nread == 3)
      send_key(buf[1],buf[2],buf[0]:
      // *** OR IF Y is REVERSED *** send_key(buf[1],(char)(-buf[2]),buf[0]); 
    }
  if(op == LE_DISCONNECT)
    return(SERVER_EXIT);
  return(SERVER_CONTINUE);
  }

```

Hitan999, 2025-03-08:
Hi, apologies for the very delayed answer. 
I tested the c version too and works (I had to use the reversed Y).

It is a bit quicked than the python version but still not fluid. Can it be because I am using a Raspberry Pi 4? Is there any other parameter that I can set to stress to see if I can get it smooth? Thank you very much!

petzval, 2025-03-09:
Download the new version 20 that was uploaded a few days ago. There are new versions of btlib.c/h and mouse.c/py. The mouse programs now read the Pi's mouse (replacing the key stroke inputs) so now do what you want without modification. The new FAST_TIMER option for set_flags allows much faster response times. The mouse operation is now smooth on my Pi4.

Hitan999, 2025-03-09:
Thank you for the new version!

I tested it and I saw a slight improvement, but still not smooth. In this case I was using a Logitech Trackman connected via PS2 to a USB Hub (it is an old device). I thought the USB HUB may be the cause. I thus tried with a USB mouse. It was slightly quicker, although not fluent. 

Not sure if it may help, but one thing that I noticed is that the speed using btferret was the speed of mouse when used in VNC Viewer. Before using the device I use ```xinput disable``` to disable it from working.
I also noticed that the program closes after disconnecting the device.

petzval, 2025-03-09:
Try speeding up the timer. It is set to 20ms, so try lower values as below. The mouse is read at this rate, so should be smooth as long as the operating system supplies data to /dev/input/mouse0 at this rate or faster.

mouse.py code:

```
btfpy.Set_flags(btfpy.FAST_TIMER,btfpy.FLAG_ON)
btfpy.Le_server(lecallback,20)  # 20ms fast timer - reduce this value
```

Hitan999, 2025-03-09:
I changed from 20 to 1 but there was no difference. Is it therefore the Pi4 reading it as longer time intervals?

petzval, 2025-03-10:
The new mouse.c/py programs run OK on my Pi4. It may be the connecting device. It should set the LE connection interval, and if this is too long it would cause the problem. See le_interval() to read the valiue and the set_le_interval_update or set_le_interval_server functions to change the interval from the Pi, or find some way of changing it from the client.