petzval/btferret #59: Init fail on second run

ehstevo, 2025-01-14:
I am using btfpy to connect to a BLE device and read notifications. On the first run everything works fine, but after I disconnect from the device once, running the script again produces the errors: 

Unable to read local board address
Bluetooth may be locked - reboot

**** initblue() FAILED ****

I am using Ubuntu 22 and have tried many things including shutting down and even removing bluez.

petzval, 2025-01-18:
If you are calling close_all and then init_blue a second time in the same program, it will not work with the current version, but may be added as a possibility for the next version.
If you are exiting the program to the command line and then re-running the code, it may be that Ubuntu is blocking Bluetooth. See section 2.1.1 in the documentation for the rfkill unblock procedure,

ehstevo, 2025-01-21:
I am exiting the program then re-running the code. I have already done the rfkill unblock procedure as well. 

I should mention it is not always the second time I run the code, sometimes it is the 3rd, 4th, or even 5th. It is not a big deal for the project I am working on to have to restart (as I most likely will be restarting the system between runs anyways) so if it is something complicated with Ubuntu or my machine no worries, but figured I'd see if you had an answer.

Thank you for the response and for the well documented repo.