petzval/btferret #62: BLE advertising name length

gregoiregentil, 2025-05-27:
In theory, it seems that BLE can advertise a name length of 29 bytes.

I'm calling init_blue, le_server with DEVICE=12345678901234567890 TYPE=mesh.

At https://github.com/petzval/btferret/blob/main/btlib.c#L2614, there is:

```
maxm = 27 - count;
```

If addname is called with flag == 0, count = 9, the name length becomes 18 characters. Is it possible to change that situation and have a longer name?

petzval, 2025-05-28:
There are two solutions. One is "legal" and gives a name length of 20 bytes. The other is a hack and gives a length of 27 bytes.  I cannot be sure it is safe - at your own risk. 

1. Reduce the mesh data length to 1. Name length = 20 bytes.
```
unsigned char c;
c = 0;
write_mesh(&c,1);
```
2. Set count to zero just before the maxm instruction. Name length = 27 bytes.
```
count = 0;
maxm = 27 - count;
```

closed 2025-05-29 by gregoiregentil

petzval, 2025-05-29:
Setting count=0 disables the mesh functions and uses the mesh section of advertising for the name. It is probably OK as long as no mesh writes are called.