petzval/btferret #64: enable only BLE (for possible lower power consumption)

fanoush, 2025-05-29:
When bluetooth is enabled on Pico (2)W the power consumption basically doubles (~17 vs 35mA), however in the datasheet 

https://www.mouser.com/datasheet/2/196/Infineon_CYW43439_DataSheet_v03_00_EN-3074791.pdf#page=76

[image]

It can be seen that classic bluetooth power draw is in tenths of mA while BLE is only ~200uA

I was trying to enable only LE and disable classic bluetooth but maybe this is currently not supported?

The `init_blue_ex` code is always registering obex serial here
https://github.com/petzval/btferret/blob/c24fe7e3570d6a6e1c6b2e149da4893f062c03c4/picostack/btlibp.c#L1451
also the clflag here https://github.com/petzval/btferret/blob/c24fe7e3570d6a6e1c6b2e149da4893f062c03c4/picostack/btlibp.c#L1605 is enabled if 'DEVICE' is present. The code is quite hard to read for me so I couldn't find whether all this is relevant and how to make sure only LE is enabled.

At least with the [mycodepico.c](https://github.com/petzval/btferret/blob/c24fe7e3570d6a6e1c6b2e149da4893f062c03c4/picostack/mycodepico.c) example the device when scanned over BLE does say it is both LE and CLASSIC device

So is there a way to enable only BLE without classic (or mesh)?

fanoush, 2025-05-29:
Thanks, I will try the LE only advertising to see if it makes any difference.

> Maybe the power use is principally in the code that is waiting for Bluetooth input - readbytes() in picostack.c. There is a sleep_ms() call in there for this purpose. I assumed that this shuts down power use while running, but maybe it doesn't.

It is the same with original pico sdk bluetooth too, I guess it is because of CYW43439 being powered on and the firmware is loaded into it. sleep_ms is run by the main RP2350/2040 CPU and that is already in the other half of power draw that is there also when CYW43439  is not used.

So either wifi is not properly disabled - but you don't use it at all in your project, you have `target_link_libraries(picostack pico_cyw43_arch_none)` there, so I guess you cannot do more. Or the bluetooth runs the classic stack too or maybe it is the SPI link between them or maybe there are some power saving/sleep hci commands that could be send to put it to lower power?

Or maybe it is feature of monolithic cyw43 driver that will load both the wifi and bluetooth firmware into the chip and the 'WLAN_REG_ON = low' condition from the datasheet is not true and the wifi core still runs. In such case I guess it would be for the Raspberry guys.

fanoush, 2025-05-29:
> When acting as an LE device, the power use is determined by the interval setting:
> 
> ```
> set_le_interval(min,max);
> set_le_interval_update(node,min,max);
> set_le_interval_server(node,min,max);
> ```
> 
> See section 4.2.54 in the README.

Oh, this is the connection interval, that is only effective after connection. How do I set the LE advertising interval? Couldn't find that in the readme. Advertising interval is not min/max, it should be one value.

petzval, 2025-05-29:
Setting the advertising interval is a feature I plan to add. At the moment it is hard coded to a fixed value of 320ms. Change it as follows:

```
Line 656 in btlib.c

unsigned char leadparam[] =
unsigned char leadparamx[] =

The two-byte values at [8] and [10] = 0200       Multiply by 0.625ms = 320ms
```